kendryte/kendryte-standalone-sdk #10: ai demo error

wipping, 2018-10-20:
In standalone sdkv0.4.0, machine can't aceess struct ai_reg_t
Missing feature: A ai demo of the standalone sdk v0.4.0
Justification:the ai demo of software can't work on sdk v0.4.0

zzxcanaan, 2018-10-20:
> In standalone sdkv0.4.0, machine can't aceess struct ai_reg_t
> Missing feature: A ai demo of the standalone sdk v0.4.0
> Justification:the ai demo of software can't work on sdk v0.4.0

Can you provide the code ? I need more detailed information.

zzxcanaan, 2018-10-20:
You shold set PLL1 frequency using sysctl_pll_set_freq(SYSCTL_PLL1, 160000000UL) if you want to use AI.
PLL1 and PLL2 is not enable by default.

wipping, 2018-10-21:
Thanks for your answer,your solution can work on sdkv0.4.0，but the box can't appear in people face。The box always appears on the left side of the screen，It's x x coordinate can't be change .The x1 always equal 0,x2 alway equal 53.The y coordinate is right

wipping, 2018-10-21:
Sorry,I made a mistake just now.The error has nothing to do with the compile option.Let me figure out the situation and then tell you again.

closed 2018-10-26 by zzxcanaan