kendryte/kendryte-standalone-sdk #36: UART hardware flow control

michprev, 2018-12-27:
https://github.com/kendryte/kendryte-standalone-sdk/blob/28a0399c65980d4314adaf3d26d89600f306d12e/lib/drivers/include/uart.h#L30-L32

But english datasheet (v0.1.5) says:

> UART1, UART2 and UART3 are general purpose UARTs and support asynchronous communication (RS232/RS485/IRDA), baud rate up to 5Mbps, **hardware flow control using CTS/RTS or (XON/XOFF)**.

Please clarify this.

ly0, 2018-12-27:
UARTH0 doesn't support hardware flow control, but UART1 2 3 support it.
Sorry it's a typo in the comment.

michprev, 2018-12-27:
The comment was perhaps meant to say that the driver does not support HW flow control yet.

michprev, 2019-01-21:
Are you sure that UART 1, 2, 3 support HW flow control? bit 4 of CPR register is set to 0 which means that AFCE_MODE is disabled.

closed 2019-04-07 by ly0