kendryte/kendryte-standalone-sdk #49: malloc was used in driver

manageryzy, 2019-01-11:
https://github.com/kendryte/kendryte-standalone-sdk/blob/8c6b3f22910fb22e215b01094fc29af0fb0288f8/lib/drivers/spi.c#L304

https://github.com/kendryte/kendryte-standalone-sdk/blob/0cd32aef17b205ac200d471f038813416eb0e9df/lib/drivers/i2c.c#L159

https://github.com/kendryte/kendryte-standalone-sdk/blob/a16b05c55f30d48c97e379a465b3426cf9d25992/lib/drivers/kpu.c#L173

https://github.com/kendryte/kendryte-standalone-sdk/blob/f43328fedf4adaca71725108c30e29b21703c91b/lib/drivers/i2s.c#L479

malloc is unsafe for dirver!!!!

michprev, 2019-03-02:
Agree, dynamic alloc should not be used in peripheral drivers.

vowstar, 2019-03-03:
Agree

swedishhat, 2019-05-25:
What kinds of issues are presented when using malloc in a peripheral driver?

smuehlst, 2020-01-06:
There may more subtle probems be caused by using malloc() in a driver, but the most fundamental problem in the source locations pointed out by @manageryzy is that the return value of malloc() is not checked for NULL. This may cause crashes or other problems in out-of-memory situtations.