kendryte/kendryte-standalone-sdk #65: cmake error

vusaalab, 2019-03-19:
When I run cmake command I am getting path below error.

- [ ] BUG REPORT

-------------------------------------------------------------------------------
> BUG REPORT
###  Document says run run below command
`~/kendryte-standalone-sdk/build$ cmake ../.. -DPROJ=testpro -DTOOLCHAIN=/opt/kendryte-toolchain/bin && make
`

Below are the errors I am getting.

```
PROJ = testpro
-- Check for RISCV toolchain ...
-- Using /opt/kendryte-toolchain/bin RISCV toolchain
SOURCE_FILES=
CMake Error at cmake/executable.cmake:11 (add_executable):
  add_executable called with incorrect number of arguments
Call Stack (most recent call first):
  CMakeLists.txt:30 (include)

CMake Error at cmake/executable.cmake:14 (set_target_properties):
  set_target_properties Can not find target to add properties to: testpro
Call Stack (most recent call first):
  CMakeLists.txt:30 (include)

CMake Error at cmake/executable.cmake:16 (target_link_libraries):
  Cannot specify link libraries for target "testpro" which is not built by
  this project.
Call Stack (most recent call first):
  CMakeLists.txt:30 (include)

```

### System

Ubuntu 16.04

zzxcanaan, 2019-03-20:
cmake .. -DPROJ=testpro -DTOOLCHAIN=/opt/kendryte-toolchain/bin && make
NOT
cmake ../..

closed 2019-03-21 by zzxcanaan

chegewara, 2019-07-14:
Hi,
i am working with make, but i am newbie with cmake.
I have prepared toolchain, but building with cmake does not work because `SOURCE_FILES=` is empty, which seems to be empty because cmake does not recognize `add_source_files` or is not setting variable with this command.

I did check cmake documentation but i cant find `add_source_files` in it:
https://cmake.org/cmake/help/v3.10/manual/cmake-commands.7.html#project-commands

Any help will be appreciated.

GongT, 2019-07-22:
Whats your project and build command looks like?

You can try to add some `MESSAGE("XXX=$XXX")` to debug cmake.

chegewara, 2019-07-22:
@GongT I think i found problem, ive read somewhere that ubuntu 18 is not compatible and i should use ubuntu 16.

PR #68 (2019-03-21): V0.5.4
Fixes #65 
Fixes #59  
Fixes #56 

- [x] I have thoroughly tested my contribution.
- [x] The code I submitted has no copyright issues.

- Major change
  - Modify KPU driver for NNCASE
  - Add APU driver
  - Add support for new toolchain
  - UART use shadow regs
  - Add spi slave driver
  - Add travis CI script

- Non-breaking bug fixes
  - Fix float issues

- Breaking change
  - Fix bus reset problem