kendryte/kendryte-standalone-sdk #73: UARTHS send big size data lost data

Neutree, 2019-04-13:
https://github.com/kendryte/kendryte-standalone-sdk/blob/78d98ae932df196579e1b9541b07435463d5d15d/lib/drivers/uarths.c#L117-L126

 When send big size data like 3K may lose data,

zzxcanaan, 2019-04-24:
Lost data when baudrate 2M on KD233 board, because the ch340 support maximum 2M.
Please tell me your operating environment and test code.

Neutree, 2019-04-24:
```c

#include <stdio.h>
#include "fpioa.h"
#include <string.h>
#include "uart.h"
#include "gpiohs.h"
#include "sysctl.h"
#include "sleep.h"

#define UART_NUM    UART_DEVICE_1

uint8_t jpeg[] ={0xFF, 0xD8, 0xFF, 0xE0, 0x00, 0x10, 0x4A, 0x46, 0x49, 0x46, 0x00, 0x01, 0x01, 0x00, 0x00, 0x01, 0x00, 0x01, 0x00, 0x00, 0xFF, 0xDB, 0x00, 0x84, 0x00, 0x16, 0x0F, 0x11, 0x13, 0x11, 0x0E, 0x16, 0x13, 0x12, 0x13, 0x19, 0x17, 0x16, 0x1A, 0x21, 0x37, 0x24, 0x21, 0x1E, 0x1E, 0x21, 0x44, 0x30, 0x33, 0x28, 0x37, 0x50, 0x46, 0x54, 0x53, 0x4F, 0x46, 0x4D, 0x4C, 0x58, 0x63, 0x7F, 0x6C, 0x58, 0x5E, 0x78, 0x5F, 0x4C, 0x4D, 0x6E, 0x96, 0x70, 0x78, 0x83, 0x87, 0x8E, 0x90, 0x8E, 0x56, 0x6A, 0x9C, 0xA7, 0x9B, 0x8A, 0xA6, 0x7F, 0x8B, 0x8E, 0x89, 0x01, 0x17, 0x19, 0x19, 0x21, 0x1D, 0x21, 0x41, 0x24, 0x24, 0x41, 0x89, 0x5B, 0x4D, 0x5B, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0x89, 0xFF, 0xC0, 0x00, 0x11, 0x08, 0x00, 0xF0, 0x01, 0x40, 0x03, 0x01, 0x21, 0x00, 0x02, 0x11, 0x01, 0x03, 0x11, 0x01, 0xFF, 0xC4, 0x01, 0xA2, 0x00, 0x00, 0x01, 0x05, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0x0B, 0x10, 0x00, 0x02, 0x01, 0x03, 0x03, 0x02, 0x04, 0x03, 0x05, 0x05, 0x04, 0x04, 0x00, 0x00, 0x01, 0x7D, 0x01, 0x02, 0x03, 0x00, 0x04, 0x11, 0x05, 0x12, 0x21, 0x31, 0x41, 0x06, 0x13, 0x51, 0x61, 0x07, 0x22, 0x71, 0x14, 0x32, 0x81, 0x91, 0xA1, 0x08, 0x23, 0x42, 0xB1, 0xC1, 0x15, 0x52, 0xD1, 0xF0, 0x24, 0x33, 0x62, 0x72, 0x82, 0x09, 0x0A, 0x16, 0x17, 0x18, 0x19, 0x1A, 0x25, 0x26, 0x27, 0x28, 0x29, 0x2A, 0x34, 0x35, 0x36, 0x37, 0x38, 0x39, 0x3A, 0x43, 0x44, 0x45, 0x46, 0x47, 0x48, 0x49, 0x4A, 0x53, 0x54, 0x55, 0x56, 0x57, 0x58, 0x59, 0x5A, 0x63, 0x64, 0x65, 0x66, 0x67, 0x68, 0x69, 0x6A, 0x73, 0x74, 0x75, 0x76, 0x77, 0x78, 0x79, 0x7A, 0x83, 0x84, 0x85, 0x86, 0x87, 0x88, 0x89, 0x8A, 0x92, 0x93, 0x94, 0x95, 0x96, 0x97, 0x98, 0x99, 0x9A, 0xA2, 0xA3, 0xA4, 0xA5, 0xA6, 0xA7, 0xA8, 0xA9, 0xAA, 0xB2, 0xB3, 0xB4, 0xB5, 0xB6, 0xB7, 0xB8, 0xB9, 0xBA, 0xC2, 0xC3, 0xC4, 0xC5, 0xC6, 0xC7, 0xC8, 0xC9, 0xCA, 0xD2, 0xD3, 0xD4, 0xD5, 0xD6, 0xD7, 0xD8, 0xD9, 0xDA, 0xE1, 0xE2, 0xE3, 0xE4, 0xE5, 0xE6, 0xE7, 0xE8, 0xE9, 0xEA, 0xF1, 0xF2, 0xF3, 0xF4, 0xF5, 0xF6, 0xF7, 0xF8, 0xF9, 0xFA, 0x01, 0x00, 0x03, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0x0B, 0x11, 0x00, 0x02, 0x01, 0x02, 0x04, 0x04, 0x03, 0x04, 0x07, 0x05, 0x04, 0x04, 0x00, 0x01, 0x02, 0x77, 0x00, 0x01, 0x02, 0x03, 0x11, 0x04, 0x05, 0x21, 0x31, 0x06, 0x12, 0x41, 0x51, 0x07, 0x61, 0x71, 0x13, 0x22, 0x32, 0x81, 0x08, 0x14, 0x42, 0x91, 0xA1, 0xB1, 0xC1, 0x09, 0x23, 0x33, 0x52, 0xF0, 0x15, 0x62, 0x72, 0xD1, 0x0A, 0x16, 0x24, 0x34, 0xE1, 0x25, 0xF1, 0x17, 0x18, 0x19, 0x1A, 0x26, 0x27, 0x28, 0x29, 0x2A, 0x35, 0x36, 0x37, 0x38, 0x39, 0x3A, 0x43, 0x44, 0x45, 0x46, 0x47, 0x48, 0x49, 0x4A, 0x53, 0x54, 0x55, 0x56, 0x57, 0x58, 0x59, 0x5A, 0x63, 0x64, 0x65, 0x66, 0x67, 0x68, 0x69, 0x6A, 0x73, 0x74, 0x75, 0x76, 0x77, 0x78, 0x79, 0x7A, 0x82, 0x83, 0x84, 0x85, 0x86, 0x87, 0x88, 0x89, 0x8A, 0x92, 0x93, 0x94, 0x95, 0x96, 0x97, 0x98, 0x99, 0x9A, 0xA2, 0xA3, 0xA4, 0xA5, 0xA6, 0xA7, 0xA8, 0xA9, 0xAA, 0xB2, 0xB3, 0xB4, 0xB5, 0xB6, 0xB7, 0xB8, 0xB9, 0xBA, 0xC2, 0xC3, 0xC4, 0xC5, 0xC6, 0xC7, 0xC8, 0xC9, 0xCA, 0xD2, 0xD3, 0xD4, 0xD5, 0xD6, 0xD7, 0xD8, 0xD9, 0xDA, 0xE2, 0xE3, 0xE4, 0xE5, 0xE6, 0xE7, 0xE8, 0xE9, 0xEA, 0xF2, 0xF3, 0xF4, 0xF5, 0xF6, 0xF7, 0xF8, 0xF9, 0xFA, 0xFF, 0xDA, 0x00, 0x0C, 0x03, 0x01, 0x00, 0x02, 0x11, 0x03, 0x11, 0x00, 0x3F, 0x00, 0xE4, 0xE9, 0x28, 0x10, 0xB4, 0x50, 0x01, 0x45, 0x00, 0x14, 0x50, 0x01, 0x45, 0x21, 0x05, 0x14, 0xC6, 0x14, 0xB4, 0x08, 0x28, 0xA4, 0x01, 0x45, 0x00, 0x14, 0x50, 0x30, 0xA2, 0x80, 0x0A, 0x28, 0x00, 0xA4, 0xA0, 0x10, 0x51, 0x40, 0x05, 0x14, 0x00, 0x51, 0x40, 0x05, 0x14, 0x00, 0x51, 0x40, 0x05, 0x14, 0x00, 0x94, 0xB4, 0xC0, 0x28, 0xA0, 0x41, 0x45, 0x03, 0x0A, 0x29, 0x00, 0x51, 0x4C, 0x02, 0x8A, 0x04, 0x14, 0xB4, 0x86, 0x14, 0x50, 0x20, 0xA2, 0x80, 0x0A, 0x28, 0x00, 0xA0, 0x50, 0x02, 0xD2, 0x50, 0x30, 0xA2, 0x80, 0x0A, 0x28, 0x00, 0xA2, 0x80, 0x12, 0x8A, 0x04, 0x14, 0x50, 0x01, 0x45, 0x00, 0x14, 0x50, 0x30, 0xA2, 0x80, 0x12, 0x96, 0x80, 0x0A, 0x28, 0x00, 0xA2, 0x98, 0x05, 0x14, 0x00, 0x51, 0x48, 0x02, 0x96, 0x80, 0x0A, 0x29, 0x80, 0x51, 0x48, 0x41, 0x4B, 0x40, 0xC2, 0x8A, 0x04, 0x14, 0x50, 0x30, 0xA2, 0x98, 0x82, 0x8A, 0x40, 0x14, 0x94, 0x0C, 0x28, 0xA0, 0x02, 0x8A, 0x04, 0x14, 0x94, 0x0C, 0x28, 0xA0, 0x02, 0x8A, 0x04, 0x14, 0x50, 0x30, 0xA2, 0x80, 0x0A, 0x28, 0x00, 0xA2, 0x80, 0x0A, 0x28, 0x00, 0xA2, 0x80, 0x16, 0x8A, 0x00, 0x28, 0xA0, 0x41, 0x4B, 0x40, 0x05, 0x14, 0x00, 0x52, 0xD0, 0x02, 0x52, 0xD0, 0x01, 0x45, 0x03, 0x12, 0x8A, 0x18, 0x82, 0x8A, 0x00, 0x28, 0xA0, 0x02, 0x92, 0x90, 0xC2, 0x8A, 0x60, 0x14, 0x94, 0x08, 0x28, 0xA0, 0x61, 0x45, 0x00, 0x14, 0x50, 0x01, 0x45, 0x31, 0x05, 0x14, 0x86, 0x14, 0x50, 0x02, 0xD1, 0x40, 0x05, 0x14, 0x00, 0xB4, 0x94, 0x00, 0xB4, 0x62, 0x81, 0x05, 0x2D, 0x00, 0x14, 0x50, 0x01, 0x4B, 0x40, 0x05, 0x25, 0x00, 0x14, 0x62, 0x90, 0x05, 0x14, 0x00, 0x94, 0x53, 0x00, 0xA2, 0x90, 0x05, 0x25, 0x00, 0x14, 0x53, 0x00, 0xA4, 0xA0, 0x02, 0x8A, 0x00, 0x28, 0xA0, 0x02, 0x8A, 0x06, 0x14, 0x50, 0x01, 0x4B, 0x40, 0x05, 0x [... cut ...]
    fpioa_set_function(5, FUNC_UART1_TX + UART_NUM * 2);
}
int main()
{
    // printf("cpu:%d\r\n", sysctl_clock_get_freq(SYSCTL_CLOCK_CPU));
    // sleep(2);
    io_mux_init();
    plic_init();
    sysctl_enable_irq();
    dmac_init();

    uart_init(UART_NUM);
    uart_configure(UART_NUM, 2000000, 8, UART_STOP_1, UART_PARITY_NONE);
    
    uint8_t tmp;
    while(1)
    {
        int ret =  uart_receive_data(UART_NUM, &tmp, 1);
        if( ret && tmp == 0x55)
            uart_send_data_dma(UART_NUM, 4, jpeg, sizeof(jpeg));
    }
}

```

zzxcanaan, 2019-04-24:
KD233？

Neutree, 2019-04-24:
Board: sipeed maix bit or dock, 
the open-ec firmware on the sipeed maix go board have bug when baud rate too high, so ignore it!
when change the CPU clock and UART clock to correct value, it can send data correctly, so maybe it's not bug

Neutree, 2019-04-24:
for this test code,  when change baudrate to 1.5M, it works correctly, but 921600 not, maybe change the CPU frequency can do.

zzxcanaan, 2019-04-24:
Thanks, I will try it.

chegewara, 2019-07-25:
I have similar/the same issue with maixduino board. UART connection between k210 and esp32 and very simple example:
```
#include <stdio.h>
#include <string.h>
#include "gpio.h"
#include "fpioa.h"
#include "uart.h"
#include "sleep.h"

int main(void)
{
    fpioa_set_function(7, FUNC_UART3_RX);
    fpioa_set_function(6, FUNC_UART3_TX);

    uart_init ( UART_DEVICE_3 );
    uart_configure(UART_DEVICE_3, 500000, UART_BITWIDTH_8BIT, UART_STOP_1, UART_PARITY_NONE);
    char recv;

    while(1){
        while ( uart_receive_data ( UART_DEVICE_3 , &recv , 1))
        {
            printf ( "%c" , recv );
            uart_send_data ( UART_DEVICE_3 , &recv , 1);
        }
        msleep(1);
    }
}
```
Sometimes data sent from esp32 are send back(sometimes corrupted), but are not logged on k210 console. When i send `\r\n` then k210 console prints old data, just like `\r\n` was missing in last message to printf.
With baudrate 115200 is much better, so maybe its esp32 or arduino issue.