kendryte/kendryte-standalone-sdk #77: Multi core interrupt enabler and disabler modification

jimaobian, 2019-04-30:
https://github.com/kendryte/kendryte-standalone-sdk/blob/b58c24f2cd057501339354b472375ad96e7c1e6d/lib/drivers/plic.c#L76

The following code could disable the Core 1 interrupt in Core 0 context. PLZ test and merge it.

```
int plic_irq_enable(plic_irq_t irq_number)
{
    /* Check parameters */
    if (PLIC_NUM_SOURCES < irq_number || 0 > irq_number)
        return -1;
    
    for(int core_id = 0; core_id < 2; core_id++)
    {
        if (plic_instance[core_id][irq_number].callback != NULL) {
            /* Get current enable bit array by IRQ number */
            uint32_t current = plic->target_enables.target[core_id].enable[irq_number / 32];
            /* Set enable bit in enable bit array */
            current |= (uint32_t)1 << (irq_number % 32);
            /* Write back the enable bit array */
            plic->target_enables.target[core_id].enable[irq_number / 32] = current;
            return 0;
        }
    }
    return -1;
}

int plic_irq_disable(plic_irq_t irq_number)
{
    /* Check parameters */
    if (PLIC_NUM_SOURCES < irq_number || 0 > irq_number)
        return -1;
    
    for(int core_id = 0; core_id < 2; core_id++)
    {
        if (plic_instance[core_id][irq_number].callback != NULL) {
            /* Get current enable bit array by IRQ number */
            uint32_t current = plic->target_enables.target[core_id].enable[irq_number / 32];
            /* Clear enable bit in enable bit array */
            current &= ~((uint32_t)1 << (irq_number % 32));
            /* Write back the enable bit array */
            plic->target_enables.target[core_id].enable[irq_number / 32] = current;
            return 0;
        }
    }
    return -1;
}
```

zzxcanaan, 2019-05-14:
You can close the interrut of peripheral  instead of plic.

jimaobian, 2019-05-16:
Acceptable.

closed 2019-05-16 by jimaobian