kendryte/kendryte-standalone-sdk #80: SPI slave not work

Pillar1989, 2019-05-13:
- [x] BUG REPORT
- [x] FEATURE REQUEST

-------------------------------------------------------------------------------
> BUG REPORT

### Expected behavior
Spi slave only use just cs, miso, mosi, clk 4 pins
### Actual behavior
Spi slave  use just cs, mosi, clk 3 pins and some weird gpio,  only accept data, can not send data.

***The data is not actually received yet***
### Test code
```bash
#include "spi_slave.h"
#include "fpioa.h"
#include "gpiohs.h"
#include "stdio.h"

#define SPI_SLAVE_INT_PIN       18
#define SPI_SLAVE_INT_IO        4
#define SPI_SLAVE_READY_PIN     19
#define SPI_SLAVE_READY_IO      5
#define SPI_SLAVE_CS_PIN        29
#define SPI_SLAVE_CLK_PIN       28
#define SPI_SLAVE_MOSI_PIN      27
#define SPI_SLAVE_MISO_PIN      21

int spi_slave_receive_hook(void *data){
    printf("%d\n", ((spi_slave_command_t *)data)->err);
    return 0;
}

int spi_slave_init(uint8_t *data, uint32_t len){
    fpioa_set_function(SPI_SLAVE_CS_PIN, FUNC_SPI_SLAVE_SS);
    fpioa_set_function(SPI_SLAVE_CLK_PIN, FUNC_SPI_SLAVE_SCLK);
    fpioa_set_function(SPI_SLAVE_MOSI_PIN, FUNC_SPI_SLAVE_D0);
    fpioa_set_function(SPI_SLAVE_INT_PIN, FUNC_GPIOHS0 + SPI_SLAVE_INT_IO);
    fpioa_set_function(SPI_SLAVE_READY_PIN, FUNC_GPIOHS0 + SPI_SLAVE_READY_IO);
    spi_slave_config(SPI_SLAVE_INT_IO, SPI_SLAVE_READY_IO, DMAC_CHANNEL5, 8, data, len, spi_slave_receive_hook);
    return 0;
}

```

### Document version

https://github.com/kendryte/kendryte-standalone-sdk/releases/tag/V0.5.5

### Hardware

[**Grove AI HAT for Edge Computing**](https://www.seeedstudio.com/Grove-AI-HAT-for-Edge-Computing-p-4026.html)

### System

What system do you use?
Ubuntu 18.04/ windows

-------------------------------------------------------------------------------
> FEATURE REQUEST

### Missing feature
missing spi slave  send data API.
### Justification
Simple, only the standard SPI four pin, asynchronous transceiver function.
### Workarounds
If I could publish the register manual for the chip, maybe I could share this PR.

xiangbingj, 2019-05-14:
> * [x]  BUG REPORT
> * [x]  FEATURE REQUEST
> 
> > BUG REPORT
> 
> ### Expected behavior
> Spi slave only use just cs, miso, mosi, clk 4 pins
> 
> ### Actual behavior
> Spi slave use just cs, mosi, clk 3 pins and some weird gpio, only accept data, can not send data.
> 
> _**The data is not actually received yet**_
> 
> ### Test code
> ```shell
> #include "spi_slave.h"
> #include "fpioa.h"
> #include "gpiohs.h"
> #include "stdio.h"
> 
> #define SPI_SLAVE_INT_PIN       18
> #define SPI_SLAVE_INT_IO        4
> #define SPI_SLAVE_READY_PIN     19
> #define SPI_SLAVE_READY_IO      5
> #define SPI_SLAVE_CS_PIN        29
> #define SPI_SLAVE_CLK_PIN       28
> #define SPI_SLAVE_MOSI_PIN      27
> #define SPI_SLAVE_MISO_PIN      21
> 
> int spi_slave_receive_hook(void *data){
>     printf("%d\n", ((spi_slave_command_t *)data)->err);
>     return 0;
> }
> 
> int spi_slave_init(uint8_t *data, uint32_t len){
>     fpioa_set_function(SPI_SLAVE_CS_PIN, FUNC_SPI_SLAVE_SS);
>     fpioa_set_function(SPI_SLAVE_CLK_PIN, FUNC_SPI_SLAVE_SCLK);
>     fpioa_set_function(SPI_SLAVE_MOSI_PIN, FUNC_SPI_SLAVE_D0);
>     fpioa_set_function(SPI_SLAVE_INT_PIN, FUNC_GPIOHS0 + SPI_SLAVE_INT_IO);
>     fpioa_set_function(SPI_SLAVE_READY_PIN, FUNC_GPIOHS0 + SPI_SLAVE_READY_IO);
>     spi_slave_config(SPI_SLAVE_INT_IO, SPI_SLAVE_READY_IO, DMAC_CHANNEL5, 8, data, len, spi_slave_receive_hook);
>     return 0;
> }
> ```
> 
> ### Document version
> https://github.com/kendryte/kendryte-standalone-sdk/releases/tag/V0.5.5
> 
> ### Hardware
> [**Grove AI HAT for Edge Computing**](https://www.seeedstudio.com/Grove-AI-HAT-for-Edge-Computing-p-4026.html)
> 
> ### System
> What system do you use?
> Ubuntu 18.04/ windows
> 
> > FEATURE REQUEST
> 
> ### Missing feature
> missing spi slave send data API.
> 
> ### Justification
> Simple, only the standard SPI four pin, asynchronous transceiver function.
> 
> ### Workarounds
> If I could publish the register manual for the chip, maybe I could share this PR.

1. spi slave mosi and miso share the same pin.
2. You can refer to the spi slave driver code.The driver performs the operation of sending and receiving data.

closed 2019-05-14 by Pillar1989

jishujun, 2019-05-17:
solved?

Pillar1989, 2019-05-17:
No, I just don't have time to deal with it

jishujun, 2019-05-17:
> No, I just don't have time to deal with it

Oh, I also encountered the same problem.

NeuerUser, 2020-12-26:
Is this bug still existing after 1.5 years or had it been fixed?

NeuerUser, 2020-12-27:
Is this bug still existing after 1.5 years or has it been fixed?

firefoxhome, 2020-12-27:
Who cares？

NeuerUser, 2020-12-27:
> Who cares？

Well, I care. I want to use SPI slave to interface a Maix board with a raspberry pi via SPI, the rpi being the master, So, yes, for me I do care if the bug is fixed or not,