kendryte/kendryte-standalone-sdk #81: in spi communication, why not use a static array?

xelll, 2019-05-14:
https://github.com/kendryte/kendryte-standalone-sdk/blob/74bb90431969c55d83ce4fa2e6cd087a152b6153/lib/drivers/spi.c#L479-L480

use a static arry will reduce malloc

zzxcanaan, 2019-05-14:
But it takes up some memory.  How do you how many memory to use?

xelll, 2019-05-14:
> But it takes up some memory. How do you how many memory to use?

it can set as max a package use

zzxcanaan, 2019-05-14:
However, this approach wastes memory

xelll, 2019-05-14:
but K210 have 6MB memory

minux, 2019-08-31:
Having a large amount of memory doesn't mean we can waste memory. (Think about program that happens to link the spi driver but otherwise doesn't use SPI.)

That being said, the API probably should be restructured so that the client passes the necessary buffer to the API.

closed 2019-08-31 by minux