kendryte/kendryte-standalone-sdk #86: What clock source does DVP SCCB clock use?

laanwj, 2019-06-02:
In this function, it seems it is based on `APB1` clock, but at the end the actual frequency is computed from `DVP`. This is inconsistent, what is correct?
```c
uint32_t dvp_sccb_set_clk_rate(uint32_t clk_rate)
{
    uint32_t tmp;
    uint32_t v_sccb_freq = sysctl_clock_get_freq(SYSCTL_CLOCK_APB1);
    uint16_t v_period_clk_cnt = round(v_sccb_freq / clk_rate / 2.0);
    if(v_period_clk_cnt > 255)
    {
        return 0;
    }
    tmp = dvp->sccb_cfg & (~(DVP_SCCB_SCL_LCNT_MASK | DVP_SCCB_SCL_HCNT_MASK));
    tmp |= DVP_SCCB_SCL_LCNT(v_period_clk_cnt) | DVP_SCCB_SCL_HCNT(v_period_clk_cnt);
    dvp->sccb_cfg = tmp;
    return sysctl_clock_get_freq(SYSCTL_CLOCK_DVP) / (v_period_clk_cnt * 2);
}
```

minux, 2019-06-04:
the explanation is (as the code is working): both are correct because DVP
clock is itself driven by APB1 clock.
But I agree that we should fix the inconsistency in the code to reduce
confusion.

laanwj, 2019-08-06:
> the explanation is (as the code is working): both are correct because DVP
clock is itself driven by APB1 clock.

They're both sourced from ACLK, but as I see it, the DVP clock has its own divider which might be different from the APB1 one.
```c
        case SYSCTL_CLOCK_DVP:
            source = sysctl_clock_source_get_freq(SYSCTL_SOURCE_ACLK);
            result = source / (sysctl_clock_get_threshold(SYSCTL_THRESHOLD_DVP) + 1);
            break;
…
        case SYSCTL_CLOCK_APB1:
            source = sysctl_clock_source_get_freq(SYSCTL_SOURCE_ACLK);
            result = source / (sysctl_clock_get_threshold(SYSCTL_THRESHOLD_APB1) + 1);
            break;
```
So `sysctl_clock_get_freq(SYSCTL_CLOCK_DVP) / (v_period_clk_cnt * 2);` can return a different value than expected. The code is working but the return value might be wrong.

closed 2022-04-14 by laanwj