kendryte/kendryte-standalone-sdk #87: Weird variable reset behavior when using timer interrupts

GitJer, 2019-07-13:
If I run the code below on the Maix Go, the variable irq_count keeps restarting at 0 at random moments. Sometimes it reaches only into the hundreds, sometimes tens of thousands. Why is this? There is nothing in the code that should make irq_count restart at 0.

```#include <stdio.h>
#include <timer.h>
#include <unistd.h>
#include <plic.h>
#include <sysctl.h>

uint32_t dummy;
// Must use volatile variable
volatile static int irq_count;

int timer_callback(void *ctx)
{
    irq_count++;
    return 0;
}

int timer_device = 2;
int timer_channel = 2;

int main(void)
{
    // Clear IRQ count
    irq_count = 0;

    // Init Platform-Level Interrupt Controller(PLIC)
    plic_init();
    // Enable global interrupt for machine mode of RISC-V
    sysctl_enable_irq();

    // initialize the timer device
    timer_init(timer_device);
    // Set timer interval to 1ms
    timer_set_interval(timer_device, timer_channel, 1000000);
    // Set timer callback function (not single shot) method
    timer_irq_register(timer_device, timer_channel, 0, 1, timer_callback, &dummy);
    // Enable timer
    timer_set_enable(timer_device, timer_channel, 1);

    while (1)
    {
        printf("%d\n", irq_count);
    }
    return 0;
}

MrYounashi, 2019-07-14:
I tried reproducing this on my KD233 and it counted up to 2,000,000 without an issue, so I don't think the code is an issue.

GitJer, 2019-07-14:
Thank you for testing this. I did notice some changes in the random restarts when putting a sleep in the while (or only printing if irq_count crosses a certain number). I do not know what causes this, but maybe the printf function and the speed of the serial connection have something to do with it.
Anyway, I used it without the printf and it seems to run fine.

closed 2019-07-14 by GitJer