kendryte/kendryte-standalone-sdk #92: V0.5.6 complier error

bugaog, 2019-08-15:
ubuntu16.04 编译V0.5.6之前的版本没问题；
编译V0.5.6版本报错如下：
[  7%] Building C object lib/CMakeFiles/kendryte.dir/bsp/entry.c.obj
[  7%] Building C object lib/CMakeFiles/kendryte.dir/bsp/entry_user.c.obj
[  8%] Building C object lib/CMakeFiles/kendryte.dir/bsp/sleep.c.obj
[  8%] Building C object lib/CMakeFiles/kendryte.dir/bsp/locks.c.obj
[  9%] Building C object lib/CMakeFiles/kendryte.dir/bsp/syscalls.c.obj
[  9%] Building C object lib/CMakeFiles/kendryte.dir/bsp/interrupt.c.obj
[ 10%] Building C object lib/CMakeFiles/kendryte.dir/bsp/printf.c.obj
[ 10%] Building C object lib/CMakeFiles/kendryte.dir/drivers/utils.c.obj
In file included from /home/tration/work/sdk/kendryte-standalone-sdk-develop/lib/drivers/utils.c:17:
/home/tration/work/sdk/kendryte-standalone-sdk-develop/lib/nncase/include/kernels/utils.h:16:10: fatal error: algorithm: No such file or directory
 #include <algorithm>
          ^~~~~~~~~~~
compilation terminated.
lib/CMakeFiles/kendryte.dir/build.make:230: recipe for target 'lib/CMakeFiles/kendryte.dir/drivers/utils.c.obj' failed
make[2]: *** [lib/CMakeFiles/kendryte.dir/drivers/utils.c.obj] Error 1
CMakeFiles/Makefile2:126: recipe for target 'lib/CMakeFiles/kendryte.dir/all' failed
make[1]: *** [lib/CMakeFiles/kendryte.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
依赖库也有安装了，可能是编译cpp的时候没有找到相应头文件吗，还是c和c++编译兼容的问题？
以下是g++和gcc的版本：
g++ --version：
g++ (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE

gcc --version:
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
有建议处理这个问题吗？谢谢！！

minux, 2019-08-15:
lib/nncase/include/kernels 和 lib/drivers/include 都包含一个 utils.h
然后编译 drivers/utils.c 的错误地用了nncase里的 C++ 头文件。

最好的办法是一个头文件改名字吧。。。

sunnycase, 2019-08-16:
Should be fixed in https://github.com/kendryte/kendryte-standalone-sdk/commit/9f54db5e4c1372ffca30d6861cfd19fcdb2a8808.

closed 2019-08-31 by minux