kendryte/kendryte-standalone-sdk #103: while trying to build happens that `_lock_t` and relative functions are not accessible from `drivers/iomem.c`

chaostheory88, 2020-01-27:
Hi. While trying to build the SDK happens that `lib/drivers/iomem.c` makes use of `_lock_t` type and its relative functions, however the build fails since those functions and type are not accessible from there.

[ 18%] Built target nncase
Scanning dependencies of target kendryte
[ 20%] Building C object lib/CMakeFiles/kendryte.dir/drivers/iomem.c.obj
/riscv/kendryte-standalone-sdk/lib/drivers/iomem.c:21:5: error: unknown type name '_lock_t'
   21 |     _lock_t *lock;
      |     ^~~~~~~
/riscv/kendryte-standalone-sdk/lib/drivers/iomem.c:24:8: error: unknown type name '_lock_t'
   24 | static _lock_t iomem_lock;
      |        ^~~~~~~
/riscv/kendryte-standalone-sdk/lib/drivers/iomem.c: In function 'iomem_free':
/riscv/kendryte-standalone-sdk/lib/drivers/iomem.c:142:5: error: implicit declaration of function '_lock_acquire_recursive' [-Werror=implicit-function-declaration]
  142 |     _lock_acquire_recursive(malloc_cortol.lock);
      |     ^~~~~~~~~~~~~~~~~~~~~~~
/riscv/kendryte-standalone-sdk/lib/drivers/iomem.c:145:5: error: implicit declaration of function '_lock_release_recursive' [-Werror=implicit-function-declaration]
  145 |     _lock_release_recursive(malloc_cortol.lock);
      |     ^~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [lib/CMakeFiles/kendryte.dir/build.make:310: lib/CMakeFiles/kendryte.dir/drivers/iomem.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:144: lib/CMakeFiles/kendryte.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

Winnaries, 2021-04-03:
I am facing the same issue. Is there any solution yet?

tuupola, 2021-08-30:
You are most likely on the `develop` branch. Try `0.5.6` instead.

```
$ git checkout 0.5.6
```