kendryte/kendryte-standalone-sdk #116: sysctl_set_power_mode does not set power mode

najiji, 2020-10-29:
- [x ] BUG REPORT
- [ ] FEATURE REQUEST

-------------------------------------------------------------------------------
> BUG REPORT

Power banks stay at the same voltage level, no matter what is set with `sysctl_set_power_mode`. For example, power bank 6 always stays at 1.8V, power bank 2 always at 3.3V

### Expected behavior

High-state voltage of the GPIO outputs are the value set with `sysctl_set_power_mode`.

### Actual behavior

High state voltages seem unaffected by  `sysctl_set_power_mode`.

### Test code

```c
    sysctl_set_power_mode(SYSCTL_POWER_BANK2, SYSCTL_POWER_V18);

    fpioa_set_function(16, FUNC_GPIOHS16);
    gpiohs_set_drive_mode(16, GPIO_DM_OUTPUT);
    gpiohs_set_pin(16, GPIO_PV_HIGH);

```

Then use a multimeter and check voltage of pin 16. Pin 16 is at 3.3V.

```c
    sysctl_set_power_mode(SYSCTL_POWER_BANK6, SYSCTL_POWER_V33);

    fpioa_set_function(36, FUNC_GPIOHS30);
    gpiohs_set_drive_mode(30, GPIO_DM_OUTPUT);
    gpiohs_set_pin(30, GPIO_PV_HIGH);
```

Then use a multimeter and check voltage of pin 36. It is 1.8V.

### Document version

Tag 0.5.6

### Hardware

Maix Dock with M1 module

### System

PlatformIO on macOS

cmdrf, 2021-08-06:
It seems the meanings of `SYSCTL_POWER_V33` and `SYSCTL_POWER_V18` are reversed. If you set `SYSCTL_POWER_V18`, you get 3.3V, and if you set  `SYSCTL_POWER_V33`, you get 1.8V.

cmdrf, 2021-08-07:
I investigated a bit further and I'm pretty sure the I/O voltage is not field-programmable. You "choose" the I/O voltage by supplying the desired voltage to the corresponding VCC pins. Example of a Sipeed board, where you can place 0 Ohm resistors to choose the voltage:
[image]

Then you use `sysctl_set_power_mode` to tell the chip which voltage you are supplying, for whatever reason.