kendryte/kendryte-standalone-sdk #123: hello-world cmake failing on d740f55894 (libnncase issue)

SebastiaanMerckx, 2021-06-15:
I cloned and installed the toolchain as in https://github.com/kendryte/kendryte-gnu-toolchain, then cloned the SDK and tried to c build hello-world. The cmake step fails.

It seems to be caused by d740f55894 since I stepped back to parent 06a2ea71f2 which seems all good.
Errors pointing to libnncase.

```bash
cmake .. -DPROJ=hello_world -DTOOLCHAIN=/opt/kendryte-toolchain/bin
```
output:

-- Check for RISCV toolchain ...
-- Using /opt/kendryte-toolchain/bin RISCV toolchain
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/kendryte-toolchain/bin/riscv64-unknown-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/kendryte-toolchain/bin/riscv64-unknown-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/kendryte-toolchain/bin/riscv64-unknown-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at lib/nncase/v1/lib/cmake/nncaseruntime/nncaseruntimeTargets.cmake:122 (message):
  The imported target "nncaseruntime" references the file

     "/home/sebastiaan/kendryte/kendryte-standalone-sdk/lib/nncase/v1/lib/libnncase.runtime.a"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/home/sebastiaan/kendryte/kendryte-standalone-sdk/lib/nncase/v1/lib/cmake/nncaseruntime/nncaseruntimeTargets.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  lib/nncase/v1/lib/cmake/nncaseruntime/nncaseruntimeConfig.cmake:1 (include)
  lib/nncase/v1/CMakeLists.txt:3 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/sebastiaan/kendryte/kendryte-standalone-sdk/build/CMakeFiles/CMakeOutput.log".

xx5988, 2021-06-15:
I also encountered the same problem！！

sunnycase, 2021-06-16:
Missing files have been added, please pull newest code.

closed 2021-06-16 by sunnycase