kendryte/kendryte-standalone-sdk #126: How can I use the latest nncase library in kendryte-standalone-sdk ?

BOGUENSONG, 2021-09-09:
Hello guys.

I use nncase Command Line Tools 1.0.0-52d9bfa for convert my custom ONNX model

and I successfully converted my model  and test on my desktop also clear. 

but when i use latest kendryte-standalone-sdk for test in my k210 device, 
test was failed. (in kpu_run_kmodel, state become blocked )

is this problem due to a nncase version difference ? 

if so, how can i use latest nncase library in kendryte-standalone-sdk ?

Should I just wait for your updates ? 

Thank you.

sunnycase, 2021-09-09:
Unzip `runtime-k210.zip` to `sdk/lib/nncase/v1`

closed 2021-09-10 by BOGUENSONG

Qinka, 2021-11-02:
> Unzip `runtime-k210.zip` to `sdk/lib/nncase/v1`

I think you should add this "how to" into nncase's readme or document of usage